JdeRobot/RoboticsAcademy #3090: First RA exercise with O3DE simulator

jmplaza, 2025-06-04:
https://o3de.org/ https://www.youtube.com/watch?v=Zx-6VhxqLKc

So far the first prototype developed by @miniesda has found a problem running O3DE inside any docker container (such as the RADI)....

Mainak54, 2026-10-25:
Hi @jmplaza  and @Apoorvgarg-creator 

I have experience building O3DE simulation pipelines and would love to help fix this Docker issue.

Since O3DE relies heavily on Vulkan, this crash is likely caused by missing GPU passthrough or headless rendering (Xvfb/EGL) configurations inside the container. To debug it, I am building a minimal Ubuntu 22.04 Docker environment with the NVIDIA Container Toolkit to reproduce and isolate the exact error.

I will post my logs and findings here shortly. If you have any crash logs from the initial prototype, please share them!

HectorMartinezCabezas, 2026-10-27:
Hi @jmplaza, @Apoorvgarg-creator and @Mainak54,

If the Vulkan issue isn’t related to the NVIDIA Container Toolkit, it is most likely because the Vulkan ICD (Installable Client Driver) file can’t be found inside the container. Without the ICD, O3DE can’t use the GPU even if the container has access to it, and falls back to CPU rendering.

The simplest fix is to mount the host’s Vulkan ICD file (e.g. /usr/share/vulkan/icd.d/nvidia_icd.json) into the container as a volume.

If you’re interested in following the development of integrating RADI and O3DE, you can check out my thesis (TFG) and the blog I’ve been working on. The thesis covers the essentials of how a RADI version works with exercises using worlds built in O3DE, while the blog documents the development process, including the problems I ran into and how I eventually solved them.

[TFG](https://www.overleaf.com/project/69c26b8d9c4c3a79c281c34d/share#1f5e1765e20e2d63415e4cb07872759257d62725b119d5fb)
[Blog](https://roboticslaburjc.github.io/2025-tfg-hector-martinez/)

Both are written in Spanish