JdeRobot/RoboticsAcademy #3097: Start Button on the exercise didn't work

pawanw17, 2025-06-08:
The start button was not responsive in my environment..

Ubuntu 22.0

Used the second method to start RA, using the two docker run commands
 
```
docker run --hostname my-postgres --name academy_db -d    -e POSTGRES_DB=academy_db     -e POSTGRES_USER=user-dev     -e POSTGRES_PASSWORD=robotics-academy-dev     -e POSTGRES_PORT=5432     -d -p 5432:5432     jderobot/robotics-database:latest

docker run --rm -it --device /dev/dri -p 6080:6080 -p 1108:1108 -p 7163:7163 -p 7164:7164 --link academy_db jderobot/robotics-academy:latest
```

[image]

javizqh, 2025-06-09:
Can you make sure that you are using the latest RADI version (delete all other RADI images and pull the new one), because the interface does not look like that anymore 

[image]

pawanw17, 2025-06-09:
Here's the process I followed
1) Cloned latest RA git clone --recurse-submodules https://github.com/JdeRobot/RoboticsAcademy.git

2) Deleted all RA and academy_db images

3) Ran this inside RA directory
sh develop_academy.sh

javizqh, 2025-06-09:
Can you try it in another browser?

pawanw17, 2025-06-09:
Got it working in chrome's incognito mode, it seems to be a cache issue... Is there a way to force browsers to not used a cached version of RA? I think then we can avoid this in the future

[image]

javizqh, 2025-06-09:
I do not know, I think it depends on the browser