JdeRobot/RoboticsAcademy #3108: Cache stored in the browser of older Robotics Academy causes failure to launch exercises.

pawanw17, 2025-06-11:
After not using RA for a few months, upon launching, I saw an older UI, which was unresponsive and did not trigger the exercise launch.

[image]

A workaround is to clear the browser cache or use incognito mode.

Can we force the browser in some way to clear RA cache on launch, or not store the cache at all (not sure about this one)

HUDASULTANA, 2025-09-10:
I am a fresher I want to work on this issue @pawanw17 can you please assign this issue to me

codezerro, 2025-09-10:
Hi @HUDASULTANA, No need to assign. Once finished, submit a pull request.

Pulkitgarg30, 2026-01-08:
Hi! I’d like to work on this issue.

From what I understand, the problem happens when old cached frontend files are loaded after a long time, which leads to an outdated UI and broken actions like exercise launch. I also noticed this issue is referenced in a few other UI-related issues, so it might be a common root cause.

I’m currently exploring how caching or service workers are handled in Robotics Academy and will share a proposed fix once I understand the current setup better.

karttikjangid, 2026-01-22:
I’m investigating whether this issue still occurs with the current frontend.
I’ll test cache behavior (service workers, cache headers, local storage) and report findings before proposing a fix.

closed 2026-02-02 by javizqh