JdeRobot/RoboticsAcademy #3163: Multi-user feature

barretto-dev, 2025-07-17:
I'm working in a new feature in your project. In the original project, its create a single container that runs both django server and manager. 

With my feature, first it's create only one container, it's runs only django server, but when a user log in, its create a dedicated container for that user, it's run only the manager

The need for this feature arose because the place i work want a online plataform where students only need to enter in broswer to use it. The feature is not finished yet, but we're already testing in a local server machine

I'm creating this issue to ask if you are interesting in this feature

javizqh, 2025-07-17:
Take a look at Unibotics, the "online" version of Robotics Academy. Unibotics works with only a single docker in the users computer, the Robotics Backend

barretto-dev, 2025-07-17:
Hello javizqh. I know about Unibotics, but my job is create something more like The Construct, no need to install docker or create containers, the student only need to enter the site with browser.

javizqh, 2025-07-17:
Unibotics can also deploy remote farms so the users can connect to the Robotics Backend from these ones instead of launching it locally

barretto-dev, 2025-07-17:
Ok. Is Unibotics repository public? Because it's necessary the system to add new exercises, so i need to create a new pages for the new exercises

javizqh, 2025-07-17:
No, it is private, but we use the exercises from Robotics Academy. If you add those exercises here, we will add them in the Next release of Unibotics

jmplaza, 2025-07-17:
Hi @barretto-dev ,

nice to meet you. I'm afraid Unibotics repository is private. Unibotics robot programming website (https://unibotics.org) provides all the exercises of RoboticsAcademy (which is fully open source) and also some premium exercises, functionalities and services for the teachers. We as non-profit organization charge some small fee for using Unibotics (as a way to reach the long term sustainability of our org, I mean, for paying some RoboticsAcademy and Unibotics software professional developers).

We already support the zero installation requirement at Unibotics, just running the RoboticsBackend in a computer farm or in a computing cloud. If you are interested on exploring this for your job, feel free to contact me. 

In addition, several universities use Unibotics in a similar way, zero installation for the user, pre-installing the RoboticsBackend on the computers of their labs. This way the users at that labs just use Unibotics without any further installation from themselves.

Regarding introducing new exercises at RoboticsAcademy, feel free to discuss them before coding. Whenever a new exercise is merged its user documentation webpage has to be also ready.

 Best,

barretto-dev, 2025-07-17:
Hi @jmplaza ,

Ok, now I undestand. Thanks to you and @javizqh for the quick answers